DFRobot/DFRobot_MultiGasSensor #1: Esp32 PCIE SIMCOM7600SA Issue

adfo0, 2022-03-17:
Hello, I am using the library in a TTGO PCIE module with Modem SIMCOM7600SA. And when trying to compile the program, I get the following errors. I don't understand what it could be, I have no experience with this type of problem.

I hope you can help me, thanks 

(The error message is much longer, but it doesn't allow me to copy the whole thing here.)

`
In file included from C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:114:0:
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino: In function 'void setup_gsm()':
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\libraries\DFRobot_MultiGasSensor-main/DFRobot_MultiGasSensor.h:36:164: error: no matching function for call to 'println(const char [10], int)'
 #define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}
                                                                                                                                                                    ^
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:447:3: note: in expansion of macro 'DBG'
   DBG("Wait  ...");
   ^
In file included from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Stream.h:26:0,
                 from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:147,
                 from sketch\ST_ODO.ino.cpp:1:
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:100:12: note: candidate: size_t Print::println(unsigned char, int) <near match>
     size_t println(unsigned char, int = DEC);
            ^
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:100:12: note:   conversion of argument 1 would be ill-formed:
In file included from C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:114:0:
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\libraries\DFRobot_MultiGasSensor-main/DFRobot_MultiGasSensor.h:36:164: error: invalid conversion from 'const char*' to 'unsigned char' [-fpermissive]
 #define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}
                                                                                                                                                                    ^
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:447:3: note: in expansion of macro 'DBG'
   DBG("Wait  ...");
   ^
In file included from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Stream.h:26:0,
                 from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:147,
                 from sketch\ST_ODO.ino.cpp:1:
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:101:12: note: candidate: size_t Print::println(int, int) <near match>
     size_t println(int, int = DEC);
            ^
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:101:12: note:   conversion of argument 1 would be ill-formed:
In file included from C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:114:0:
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\libraries\DFRobot_MultiGasSensor-main/DFRobot_MultiGasSensor.h:36:164: error: invalid conversion from 'const char*' to 'int' [-fpermissive]
 #define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}
                                                                                                                                                                    ^
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:447:3: note: in expansion of macro 'DBG'
   DBG("Wait  ...");
   ^
In file included from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Stream.h:26:0,
                 from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:147,
                 from sketch\ST_ODO.ino.cpp:1:
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:102:12: note: candidate: size_t Print::println(unsigned int, int) <near match>
     size_t println(unsigned int, int = DEC);
            ^
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:102:12: note:   conversion of argument 1 would be ill-formed:
In file included from C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:114:0:
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\libraries\DFRobot_MultiGasSensor-main/DFRobot_MultiGasSensor.h:36:164: error: invalid conversion from 'const char*' to 'unsigned int' [-fpermissive]
 #define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}
                                                                                                                                                                    ^
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:447:3: note: in expansion of macro 'DBG'
   DBG("Wait  ...");
   ^
In file included from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Stream.h:26:0,
                 from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:147,
                 from sketch\ST_ODO.ino.cpp:1:
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:103:12: note: candidate: size_t Print::println(long int, int) <near match>
     size_t println(long, int = DEC);
            ^
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:103:12: note:   conversion of argument 1 would be ill-formed:
In file included from C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:114:0:
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\libraries\DFRobot_MultiGasSensor-main/DFRobot_MultiGasSensor.h:36:164: error: invalid conversion from 'const char*' to 'long int' [-fpermissive]
 #define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}
                                                                                                                                                                    ^
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:447:3: note: in expansion of macro 'DBG'
   DBG("Wait  ...");
[... 413 lines cut ...]
                                                                                                                                                                    ^
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:484:9: note: in expansion of macro 'DBG'
         DBG("Network connected");
         ^
In file included from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Stream.h:26:0,
                 from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:147,
                 from sketch\ST_ODO.ino.cpp:1:
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:104:12: note: candidate: size_t Print::println(long unsigned int, int) <near match>
     size_t println(unsigned long, int = DEC);
            ^
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:104:12: note:   conversion of argument 1 would be ill-formed:
In file included from C:\Users\adolf\OneDrive\Documents\GitHub\esp32\ST_ODO\ST_ODO.ino:114:0:
C:\Users\adolf\OneDrive\Documents\GitHub\esp32\libraries\DFRobot_MultiGasSensor-main/DFRobot_MultiGasSensor.h:36:164: error: invalid conversion from 'const char*' to 'long unsigned int' [-fpermissive]
 #define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}
[... cut ...]
In file included from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Stream.h:26:0,
                 from C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:147,
                 from sketch\ST_ODO.ino.cpp:1:
C:\Users\adolf\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Print.h:105:12: note: candidate: size_t Print::println(long long int, int) <near match>
     size_t println(long long, int = DEC);
            ^
`

adfo0, 2022-03-17:
[Issue ESP32 PCIE.txt](https://github.com/DFRobot/DFRobot_MultiGasSensor/files/8286312/Issue.ESP32.PCIE.txt)

pkx890, 2022-03-18:
Mask the code in library files:#define DBG(...) {Serial.print("[");Serial.print(__FUNCTION__); Serial.print("(): "); Serial.print(__LINE__); Serial.print(" ] 0x"); Serial.println(__VA_ARGS__,HEX);}

closed 2023-03-06 by fary99