DFRobot/DFRobot_MultiGasSensor #5: Gas type not reported after some time and wrong values

ipavelic1, 2023-04-14:
Hi I am using an ESP32 with I2C communication to read from SEN0466, SEN0470, SEN0471, SEN0472 sensors. When I power on sensors (power supply is 3.3V) all of them are detected as expected, with “queryGasType()” command. However, after some time, I noticed that “queryGasType()” command stop reporting type of sensors and then value start to be incorrect. One more thing that I notice is that this happening for about every 4 hours. When I power on sensors first 4 hour works I believe OK, all sensors are reported properly with “queryGasType()” command. After first 4 hours of working begin with that first two sensors didn’t report (NO2 and O3, and for the values they give 11265ppm for NO2 and 298 ppm for O3). In next 4 hour next two sensors stop reporting “queryGasType()” and now we have situation where all 4 sensors stop working and for the values they give next CO 4ppm, NO2 0ppm, SO2 43 ppm and O3 10753ppm. In next 4 hour situation is not change with reporting type of sensor with “queryGasType()” command but the values have change to CO 1024ppm, NO2 0ppm, SO2 11009ppm and O3 10753ppm. So, I can conclude that this is not a problem with I2C communication, especially because I have some other device on I2C communication that work just fine for all of that time. I didn’t write that I use “PASSIVITY” mode in this situation, and the library version is latest 2.0. I also read and “gasO3.getSensorVoltage()” and this value is same as on pin VO (measured with universal instrument UNI-T UT61E). 
In meanwhile I edit your library to get answer what code number (HEX) get in case when don’t recognize sensor on “queryGasType()” command. So I get for SO2  1

I was try in "PASSIVITY" and "INITIATIVE" mode all the same.

And the error look like this:
14:00:51.717 -> Gas type: SO2
14:00:51.717 -> Gas SO2 concentration is: 0.00 ppm
14:00:51.717 -> The board gas 3 temperature is: 24.08 °C
14:00:51.806 -> Sensor SO2 output voltage is: 0.55 V
14:00:51.901 -> Sensor NO2 calculation from voltage is: -0.55 ppm 

14:01:44.034 -> Gas type: 1
14:01:44.034 -> Gas SO2 concentration is: 43.00 ppm
14:01:44.034 -> The board gas 3 temperature is: 24.08 °C
14:01:44.134 -> Sensor SO2 output voltage is: 0.55 V
14:01:44.224 -> Sensor NO2 calculation from voltage is: -0.61 ppm

On wiki you have one command “gas.setGasType(/*Gas type*/gas.O2);” that dose not exist in this version of library (I don’t know for prevised) so maybe you can update the wiki page of this sensors.

BelKed, 2023-07-05:
I had a similar problem while I tested the SO2 and CO sensors.
Have anyone managed to solve it?

96liuzhixin, 2023-07-31:
We have solved this problem in the new firmware version, if you need it, please contact the after-sales service

closed 2023-07-31 by 96liuzhixin