DFRobot/DFRobot_MultiGasSensor #9: Supress warnnings on some architectures

hpsaturn, 2023-08-08:
Hi,

Thanks in advance to give support to this library. We are using this library in [CanAirIO Sensors layer](https://github.com/kike-canaries/canairio_sensorlib) and we are noticed that we have some warnings on some architectures.  

## Esp82666

```cpp
.pio/libdeps/esp8266/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp: In function 'void analysisAllData()':
.pio/libdeps/esp8266/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:143:16: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
  143 |           else if ((_temp > 20) && (_temp >= 50))
      |                ^~
.pio/libdeps/esp8266/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:145:13: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  145 |             break;
      |             ^~~~~
.pio/libdeps/esp8266/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp: In member function 'virtual int16_t DFRobot_GAS_HardWareUart::readData(uint8_t, uint8_t*, uint8_t)':
.pio/libdeps/esp8266/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:767:7: warning: unused variable 'i' [-Wunused-variable]
  767 |   int i = 0;
      |       ^
.pio/libdeps/esp8266/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:768:25: warning: 'length' may be used uninitialized in this function [-Wmaybe-uninitialized]
  768 |   for (int i = Reg; i < length; i++)
```

## Atmelsam

```cpp
.pio/libdeps/atmelsam/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp: In function 'void analysisAllData()':
.pio/libdeps/atmelsam/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:143:16: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
           else if ((_temp > 20) && (_temp >= 50))
                ^~
.pio/libdeps/atmelsam/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:145:13: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
             break;
             ^~~~~
.pio/libdeps/atmelsam/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp: In member function 'virtual int16_t DFRobot_GAS_HardWareUart::readData(uint8_t, uint8_t*, uint8_t)':
.pio/libdeps/atmelsam/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:767:7: warning: unused variable 'i' [-Wunused-variable]
   int i = 0;
       ^
.pio/libdeps/atmelsam/DFRobot_MultiGasSensor/DFRobot_MultiGasSensor.cpp:768:23: warning: 'length' may be used uninitialized in this function [-Wmaybe-uninitialized]
   for (int i = Reg; i < length; i++)
                     ~~^~~~~~~~
```

96liuzhixin, 2023-08-10:
We are about to modify the warning for line 767, but we cannot find the warning for line 143

hpsaturn, 2023-08-15:
Maybe you able to replicate the warnning using PlatformIO like this:

```bash
git clone https://github.com/kike-canaries/canairio_sensorlib.git &&
cd canairio_sensorlib && pio run
```

96liuzhixin, 2023-08-16:
This library is https://github.com/dfrobot/dfrobot_multigasSor.git
The board we tested
Arduino uno √
Firebeetle ESP32 √
FirebeTle ESP8266 √
Firebeetle m0 √
Leonardo √
Microbit √
Arduino mega2560 √
If you want to support more, please contact PMhttps://github.com/DFRobot/DFRobot_MultiGasSensor.git

closed 2023-08-16 by 96liuzhixin

hpsaturn, 2023-09-29:
I think that we need open again this issue. One user from the support channel of CanAirIO, reports that the current master branch of this library, the NH3 values stopped of him device. The only change was pass from the version 2.0.0 to this master (with this current fix). We returned again to the version 2.0.0 (the official published on PlatformIO) and the air quality station returned to publish the data.

We don't have the cause yet, but we believe that is something with:

[image]

96liuzhixin, 2023-10-23:
I looked at the documentation for the calibration algorithm and the algorithm provided later is more accurate